APRIL-ZJU/Gaussian-LIC #12: cuda17

bugmaker1027, 2025-07-24:
Have you encountered a  problem like the one shown in the image below?

[image]

 It seems to be an issue with the versions of CUDA and CMake. Using C++17 in the CMakeLists is affecting the compilation of CUDA. However, the method found online, which is setting set(CMAKE_CUDA_STANDARD 14) and set(CMAKE_CUDA_STANDARD_REQUIRED ON), didn't work. And I don't want to upgrade the CMake version. Has anyone tried other solutions to fix this?

Jerry-locker, 2025-07-25:
Hi, what is the version of your CMake?

bugmaker1027, 2025-07-29:
The default CMake version on Ubuntu 22.04 is 3.17, but upgrading it to 3.28 resolved this issue. thx

Jacke1ine, 2025-08-11:
我的电脑版本cuda是11.8，项目支持cuda11.8版本的吗？
项目对CMake版本是不是也有要求呀？也是需要>3.28版本吗？

Jerry-locker, 2025-08-25:
> 我的电脑版本cuda是11.8，项目支持cuda11.8版本的吗？ 项目对CMake版本是不是也有要求呀？也是需要>3.28版本吗？

使用cuda11.8版本也可以，只要准备对应版本的LibTorch即可。

miiiker109-maker, 2026-09-08:
> > Ubuntu 22.04 的默认 CMake 版本是 3.17，但升级到 3.28 后这个问题就解决了。谢谢
> 
> 你好，我想请问一下，我的电脑也是ubuntu 22.04,但是ubuntu22.04仅支持ros2,你是采用docker隔离复现的吗？

您好，请问解决了吗？