APRIL-ZJU/Gaussian-LIC #19: 关于渲染结果

lpw-png, 2025-09-03:
我在使用自己的相机雷达（mid360）时出现了静止时效果不错，但当相机运动时渲染效果变差的问题。如下所示：

[image]
[image]
[image]
我想知道这大概是因为相机与imu外参标定不准确吗

lpw-png, 2025-09-03:
是的，使用的fastlivo的硬同步方法。但是在cocolic的配置文件中，对于cam to imu的标定时，我使用kalibr标定得到的timeoffset为-0.04，我看您的原始配置文件写的是0，会是这里的影响吗

Jerry-locker, 2025-09-04:
如果 LIC 设备移动速度较快，-0.04s 的 timeoffset 对建图质量影响还是挺大的。
我不确定 Coco-LIC 当前版本的 camera.yaml 中的 time_offset 是否生效，这两天我会检查一下并更新对应的代码。

另外，可以先试着跑一下 FAST-LIVO2：1）检查静止情况下的彩色点云地图重建情况，如果可视化效果较好，说明激光-相机空间外参（rotation + translation）没问题；2）检查运动情况下的彩色点云地图重建情况，如果可视化效果变差，说明受到 timeoffset 影响了，需要补偿。